sinara-hw/RFSOC_SAWG #12: Do you have a particular use case?

gkasprow, 2021-12-06:


gkasprow, 2022-03-18:
Here I need feedback. The first version of HW would be wideband. We can also use Sayma-like AFE with an attenuator, but anyway need some feedback from possible users.

kaolpr, 2022-04-07:
Unless you have some particular use-case in mind, maybe we should consider freezing this project? At least until we know something about use-case?

gkasprow, 2022-04-07:
Let's make a basic test/characterization board. I already ordered Trenz modules. After that, we would know how capable they are in terms of analog performance.

gkasprow, 2022-07-26:
Why would you need RF part (ADC/DAC) for that?

dhslichter, 2022-07-28:
We don't have a strong need for RF ADC at present, but RF DAC for direct digital synthesis of multi-tone modulated waveforms up to ~3.5 GHz would be useful and relevant.  This would involve synthesizing Phaser-like waveforms (typical of what would be done either with old-style Phaser or the new STFT Phaser encoding).  I don't see a need for dc coupling.  ARTIQ integration would be key.  The primary rationale for using a board like this would be to reduce footprint and eliminate the need for separate modulators, DDS channels, Phasers, etc that would currently be used to generate such signals.  I would characterize us as being curious but not desperate for this kind of solution.

gkasprow, 2022-07-28:
I'm making a simple baseboard to characterize the Trenz modules and see what limits us.
Since we are migrating towards DIOT, there is already [ZynQ US system board ](https://ohwr.org/project/diot-sb-zu/wikis/home)available that can be used as the controller.

andrewvh4, 2022-07-28:
We also only need the RF DAC component, however frequencies up to ~500GHz would be sufficient for trapped ions. 

>>> Why would you need RF part (ADC/DAC) for that?

Tighter integration between RF DAC peripherals and the master controller, as well as better utilization of RFSoC fabric.

jordens, 2022-07-28:
Only 500 GHz?

gkasprow, 2022-09-07:
We identified several use cases. Some of them require low latency PCIe link, some require 100G Ethernet. I think I covered them all with current design.

closed 2022-09-07 by gkasprow