sinara-hw/RFSOC_SAWG #23: initial mechanical construction

gkasprow, 2022-09-29:
the main board 
[image]

the debugging mezzanine
[image]

kaolpr, 2022-09-29:
I believe there should be at least one more CPCIs connector?

Those fly-overs seem to be a good idea, but maybe we could trade off single-link bandwidth for number of links and use RTM with classic SFP?

gkasprow, 2022-09-30:
There will be only one CPCIS connector because the Trenz module has only 14 LVDS lines, and we need a few of them for clocking. So only 8 of them are left for communication. But we have two SFP cages that can run DRTIO links.
The Flyovers have a QSFP option as well - you simply buy a copper cable with Flyover on one side and QSFP on the other side.
The mezzanine has free space to host the QSFP connector with power and control pins. One can also use two MTP fiber connectors on the front panel. The RTM connector won't let us utilize QSFP28 signaling (100G Ethernet). We need a low latency PCIe link for hybrid quantum computing (quantum data center). We need 100G for direct RF sampling in other applications.
Once we decide to build a 3U module using the RFSOC chip, we could use 16-channel LVDS CPCIS/EEM connectivity.